HebiRobotics/hebi-scope-feedback #35: Big hang when clicking a plot button the first time.

daverolli, 2018-10-23:
In macOS when clicking a plot button for the first time there is a ~2 second hang before the plot pops up.  There is no animation from the mouse-click so it looks like the app is hanging.  It was long enough for the OSU guys to notice when doing a live demo.  

On windows the stutter is much shorter, and there's a partial animation that makes it clear the click was received.

ennerf, 2018-10-23:
Future note: There is also a ~2 second hang the first time someone enters a bad value into a text field, e.g., gains. Pre-Initializing without rendering didn't work, and initializing while rendering added 2 seconds to the startup, which was even worse.

ennerf, 2019-10-14:
This is an issue for Linux as well. May be related to initializing the communications?

ennerf, 2021-06-18:
Solved in 1.6.4

closed 2021-06-18 by ennerf