HebiRobotics/hebi-scope-feedback #38: hardware rendering on linux

iamtesch, 2018-10-23:
I may be getting the general description wrong here, but there is a bug in the software rendering on Linux that causes a significant memory leak.  This was fixed on the previous version of scope, but I don't think has been addressed on the redesign yet.

I wouldn't necessarily consider this a blocker for an initial release, but this is something that should be highest priority after initial release, as it makes Scope very frustrating to use on linux except for quick tasks, and can lead to crashes and extreme slowdowns.

ennerf, 2018-10-23:
[image]

There is a switch in the settings that allows users to turn on software rendering if they have an old Mesa3D version that causes the memory leak.

ennerf, 2018-10-23:
The bug was in Linux' OpenGL drivers `Mesa3D`. The bug has been fixed in versions that ship with Ubuntu 16.04 update 4 as well as 18.04

closed 2018-10-23 by ennerf