HebiRobotics/hebi-scope-feedback #45: Change plotting framework

ennerf, 2018-10-25:
Currently, we are using [JChart2d](http://jchart2d.sourceforge.net/), which is based on the pretty much deprecated Java Swing UI toolkit. It doesn't support things like DPI scaling, but it's so far the only plotting library I found specifically for plotting `real-time` data. That being said, it still has some allocations.

At some point it'd be nice if we could create a custom real-time plotting library for JavaFX.

rLinks234, 2018-10-25:
The best way to do this is interface directly with vertex buffers, unfortunately. The best way I've been able to achieve this is using OpenGL and D3D directly, which isn't very user friendly. Most libraries which provide a user friendly wrapper around these APIs are in C and C++

ennerf, 2018-10-25:
The performance of the JavaFX renderer will be fine for this. It's a decent platform independent wrapper around OpenGL and D3D with a Software fallback. Working with `JOGL` is a bit of a pain, and I've never looked into D3D. I also still want to keep it be be styleable with CSS.

My issue with the provided JavaFX plotting libraries is just that they use a list of point objects of `Double` values, i.e., the boxed type. The allocation rate makes it unusable for displaying live data. See [Line Chart Example](https://docs.oracle.com/javafx/2/charts/line-chart.htm)

rLinks234, 2018-10-25:
Your best bet here is to use something analogous to a SurfaceView + Canvas (hardware accelerated surface) in a view which provides styling using CSS.

I've gone down this road before and know from looking at the source code that basically all common drawing APIs (e.g., Canvas) are not designed around displaying streamed data. They normally dispose of your tesellated line vertices (they get transformed from lines into triangles unless you draw ugly 1px wide lines) and sometimes do CPU _and_ GPU allocated memory per call (because they do not know that you want to draw 1000 line segments from a buffer per frame).

Depending on the toolkit, this may cause an incredible amount of useless allocations. I would watch out for this doubling memory usage from what it is already.

FWIW, I made an Android demo before which used a `canvas.drawText(coloredText, pos, paint);` approach, which struggled to run at more than 15fps with only a few hundred words (at most) on screen. Using OpenGL ES 2 directly with glyphs in a texture atlas gave me 60fps with no problem, while rendering an entire 6000LOC file (most of which was off the screen). The reasoning behind this was that the `drawText` was reallocating multiple buffers (in C code) on each call, and redoing the calculations on every single frame. Yes, this is _Android_ Java, but it _is_ OpenJDK with, quite frankly, a quite similar graphics/drawing API compared to JavaFX.

ennerf, 2019-09-18:
Snapshot switched to [ChartFx](https://github.com/GSI-CS-CO/chart-fx) and Java 13

closed 2019-10-15 by ennerf