HebiRobotics/hebi-scope-feedback #49: Remember plot PPI settings across restarts

daverolli, 2018-11-02:
Generally once you're on a computer, you want a specific PPI.  For example, in my case both my normal computers work best with 100% instead of the default of 150%, so it has to get reset every time the app is launched.  It would be nice the setting remembered where it was set to and used that on the next load.  

This might even let us move the PPI selection to the "Settings" menu instead of having to be on the "Monitoring" Panel.

daverolli, 2018-11-02:
Would be good for other things too like what folder you load/save gains from/to.

ennerf, 2018-11-09:
Made it persistent. I often switch between different DPI scales at conferences, when I'm explaining something, and when I look at a single vs multiple plots. Unless we find a better use for the space, I'd vote for not moving it to settings.

ennerf, 2018-11-09:
Fixed in snapshot. Please confirm.

closed 2018-11-12 by ennerf