HebiRobotics/hebi-scope-feedback #50: Permanent hang in macOS when closing multiple windows

daverolli, 2018-11-02:
In macOS, if you open multiple plotting windows at once (plotting when modules are multi-selected), and you don't put each one of them into front focus, hitting 'Close All' causes Scope to hang until you force-quit.   If you click on each plotting window so that all of them see front focus at least once this doesn't happen.  If you close the plot windows one at a time, either with the mouse or with `Cmd+W`, this doesn't happen.

For example if I select 4 modules, click around or move 3 of the windows, but never touch the 4th, the app will hang when I click 'Close-All'.  This happens a lot because I usually open a bunch of plot windows, realize I have the PPI setting wrong, then click 'Close All' to try to fix.

ennerf, 2018-11-09:
Issue may be fixed in latest snapshot. Please try again.

daverolli, 2019-06-28:
Problem still occurs in macOS in latest 1.5-SNAPSHOT.

ennerf, 2019-07-02:
Found solution in https://stackoverflow.com/a/27827033. Fixed it on my test mac

closed 2019-07-02 by ennerf