HebiRobotics/hebi-scope-feedback #52: "Advanced" parameters not sent when hidden.

iamtesch, 2018-11-08:
Repro:
* Turn on "advanced mode"
* Change the "Min output" for the Position gain (or any gain)
* Turn off "advanced mode"
* Click "send"
* Turn on "advanced mode" - notice that the parameters which were hidden are not sent

(This was particularly noticeable with a "load file" and then "send", where the values of the file are not actually on the module as expected after this operation)

ennerf, 2018-11-09:
It's not a bug, it's a feature!

I don't think it should send anything that the user isn't aware of. I was planning on making this more obvious with #3.

Do you think that'd work well enough?

ennerf, 2018-11-09:
After further discussion, we will probably change it to send hidden fields as well.

ennerf, 2018-11-09:
Should be fixed in 1.0.1-SNAPSHOT. Please confirm

daverolli, 2018-11-09:
Confirmed that it works in macOS / Windows

closed 2018-11-12 by ennerf