HebiRobotics/hebi-scope-feedback #68: Scope - linux performance

iamtesch, 2019-03-26:
At a recent customer visit, running Scope froze the customer's system and made it unresponsive (on Ubuntu 16.04).  Is there any way to limit the memory usage or CPU usage to prevent such issues?  Or somehow optimize the usage of system resources more?

I recognize this is more of a long term issue, I just want to make sure we have an issue open so we can track discussion/thoughts as to this, and document progress.

ennerf, 2019-04-05:
As Daniel mentioned, this is probably related the bug in the MESA driver on older versions of Linux. See #38 for details.

There is a switch in the settings dialog to enable software rendering to work around this memory leak issue. Please check with the customer and see if it still happens after enabling software rendering and restarting Scope.

rLinks234, 2019-04-05:
I actually didn't think of this - you can download newer versions of `mesa` through ppa's. That is another suggested workaround on older/slower computers which may not like having to do software rendering. I'd be willing to type up a small piece of documentation to suggest this if you would find it to be useful, @ennerf ?

ennerf, 2019-05-15:
Another note that may result in some improvements. JavaFX 8 linked against GTK2 by default, and JavaFX 11 changed it to GTK3. However, there are some flags to force GTK3 in older versions as well.

https://stackoverflow.com/a/22457177

Add flags to `Scope.cfg`

[image]

I didn't notice anything immediately obvious on my test machine, but I haven't done much testing yet.

closed 2019-10-15 by ennerf