HebiRobotics/hebi-scope-feedback #75: Scope - persist buttons + toggles

iamtesch, 2019-06-10:
Issues that came up from customer feedback today
- [ ] The "persist" button on the dashboard should probably be removed; it is unintuitive because there is nothing on this pane which doesn't already persist when it is set
- [ ] The 'persist sends' toggle state should probably not be saved between reboots.  This makes it easy to accidentally persist sends the next time you have the GUI open, especially if you just toggled this as you were initially exploring the GUI.

ennerf, 2019-06-17:
Removed the persist button from Dashboard. The idea was that users should be able to persist without having to go to Gains and send a dummy message with the persist flag on. Considering that (1) many things auto-persist now, and (2) we are moving away from persisted gains towards loading them each run I think removing it makes sense

ennerf, 2019-06-17:
Removed persisting behavior from the dashboard and monitoring toggle

closed 2019-06-17 by ennerf