HebiRobotics/hebi-scope-feedback #78: Module state

iamtesch, 2019-06-10:
Issues that came up from customer feedback today

- Previous versions of Scope had the current LED color visible somewhere in the GUI.  This was a useful feature for quickly checking the general state of each module without looking at the LEDs individually, or reading through all the module state on each one
- The "Locked" status field could probably be renamed to something a little more clear -- potentially "command lockout state", or something else

ennerf, 2019-06-17:
* I always thought the LED color mapping looks a bit weird. Is there any state that can't be determined by a quick look at the Dashboard status overview? aside from searching-for-ip, which we can't display in any case.
* renamed status to "COMMAND LOCK"

iamtesch, 2019-06-17:
The dashboard status overview technically gives everything except for a manually set color (e.g., if this was set from a different piece of code, or from the "set LED red" button in scope).  However, it is still a lot more visual state to parse -- you are going through a lot of individual text fields instead of one LED image if you are, say, scanning through each module.

@daverolli had asked if we could have this back

daverolli, 2019-06-17:
There are a couple uses.  In one case when you set LED color, either from Scope or in the API, you can't tab thru the modules and see which ones are set vs cleared.  There are some internal states that don't show up, like actuators where effort isn't calibrated.

closed 2019-07-02 by ennerf