HebiRobotics/hebi-scope-feedback #80: Issues with touch screen on screen keyboard on Ubuntu

curtishebi, 2019-06-13:
I am using a 1280x800 touch screen with Ubuntu 18.04. There are a few issues that all seem related to the on screen keyboard. First of all, if I click any of the plotting buttons it will pop up the plot but also pop up the keyboard. This will sometimes cause the windows to shift up - if they do they will not return to their correct position.

Also, it is very easy to click on the large module name text - which is editable and then will pop up the onscreen keyboard, which will sometimes move the windows up. I had no idea this was editable in the first place - it may be better to just make it static. Sometimes after clicking this text, if I click away and then click back somewhere else it will still pop up the keyboard.

Lastly, sometimes after the keyboard pops up and the windows shift around for a bit, I can get it into a state where the place that I touch on the scope window is offset from where the clicking actually happens. It is very weird.

This screen is getting used for the Chevron bendy arm and the inspection arm we will be building later in the summer. I can send you one if that would help!

ennerf, 2019-06-17:
the editable family/name were removed. will look into other issues when I get the test display

ennerf, 2023-02-01:
Most of these have been fixed and afaik we are not using those screens anymore.