HebiRobotics/hebi-scope-feedback #82: Keeping plot open consumes endless memory

rLinks234, 2019-06-17:
I've noticed this before when having a plot open for an extended period of time (tens of minutes to hours), which I imagine may be a common thing for users.

After about 3 hours of Scope having only 1 plot open (position plot of some module not moving at all), Scope is utilizing 1,678 MiB of **resident** (not virtual) memory. I frequently checked the memory usage, along with the `/proc/<pid>/smaps` file, which prints out the allocated pages to the process. Below shows the relevant pages of interest:

Note: I ran `cat smaps | grep -B4 -e "^Rss"` to filter out some of the irrelevant fields.
```
00400000-00402000 r-xp 00000000 08:00 25428074                           /mnt/develop/develop/hebi/Scope/Scope
Size:                  8 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                   8 kB    <-- OK - also a mmap'd file so irrelevant
--
00601000-00602000 r--p 00001000 08:00 25428074                           /mnt/develop/develop/hebi/Scope/Scope
Size:                  4 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                   4 kB    <-- OK - also a mmap'd file so irrelevant
--
00602000-00603000 rw-p 00002000 08:00 25428074                           /mnt/develop/develop/hebi/Scope/Scope
Size:                  4 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                   4 kB    <-- OK - also a mmap'd file so irrelevant
--
01375000-015ec000 rw-p 00000000 00:00 0                                  [heap]
Size:               2524 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                2280 kB    <-- the "heap" as determined by sbrk
--
6ca800000-6dee80000 rw-p 00000000 00:00 0 
Size:             334336 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:              237320 kB    <-- Problem number 1. 231 MiB
--
6dee80000-76e300000 ---p 00000000 00:00 0 
Size:            2347520 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                   0 kB    <-- NOTE: this is probably a mmap'd page by the JVM to catch when the GC arenas reach the maximum size and have to be enlarged to fix more data
--
76e300000-7c0000000 rw-p 00000000 00:00 0 
Size:            1340416 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:             1340200 kB    <-- Problem number 2. 1.2781 GiB

...
```

Problem 1 and 2 **scale in size depending on `-XmsNNNN -XmxNNNN` parameters**.

For instance, on a "fresh" run of Scope with `-Xms8192m -Xmx10000m`, the above smaps becomes (immediately upon app opening):

```
00400000-00402000 r-xp 00000000 08:00 25428074                           /mnt/develop/develop/hebi/Scope/Scope
Size:                  8 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                   8 kB
--
00601000-00602000 r--p 00001000 08:00 25428074                           /mnt/develop/develop/hebi/Scope/Scope
Size:                  4 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                   4 kB
--
00602000-00603000 rw-p 00002000 08:00 25428074                           /mnt/develop/develop/hebi/Scope/Scope
Size:                  4 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                   4 kB
--
012eb000-0153b000 rw-p 00000000 00:00 0                                  [heap]
Size:               2368 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                2088 kB
--
54f000000-6d2f00000 rw-p 00000000 00:00 0 
Size:            6355968 kB    <--- 6.06 GiB "allocated"
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:               54260 kB    <--- Same map as Problem 1 above: Only ~50MiB actually mapped (being used)
--
6d2f00000-6efb00000 ---p 00000000 00:00 0 
Size:             471040 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                   0 kB
--
6efb00000-7b6e80000 rw-p 00000000 00:00 0 
Size:            3264000 kB   <--- 3.11 GiB "allocated"
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:              283424 kB    <--- Same map as Problem 2 above: Surprisingly ~275 MiB being mapped by default?
--
7b6e80000-7c0000000 ---p 00000000 00:00 0 
Size:             148992 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                   0 kB
--
7c0000000-7c0520000 rw-p 00000000 00:00 0 
Size:               5248 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Rss:                5116 kB
```

This is **definitely** not an i915 memory leak in MESA, as `glxinfo` reports
```
OpenGL core profile version string: 4.5 (Core Profile) Mesa 18.3.6
```
and these issues were fixed as I [reported here](https://github.com/HebiRobotics/hebi-scope-feedback/issues/73#issuecomment-488318868).

I am using the JVM which comes packaged with the Scope download for Linux_x86_64 on docs.hebi.us .

Probably not relevant, but to mention the system:

```
-------------------------- 
OS: Fedora release 29 (Twenty Nine) x86_64 
Kernel: 5.0.14-200.fc29.x86_64 
Shell: bash 4.4.23 
Resolution: 1920x1200
DE: GNOME 3.30.2 
WM: GNOME Shell 
Terminal: gnome-terminal 
CPU: Intel i7-8700K (12) @ 4.700GHz 
GPU: Intel UHD Graphics 630 
Memory: 7817MiB / 15711MiB 
```

ennerf, 2019-09-05:
It looks like it's fixed in OpenJFX 12

rLinks234, 2019-09-05:
I still need to re-run this before confirming

closed 2019-12-19 by ennerf