HebiRobotics/hebi-scope-feedback #85: [Linux] Module discovery

Giggus, 2019-08-09:
Hi, 

I just received my new pc with Ubuntu 18.04. 
My java version is openjdk 11.0.4. 

When I plug the robot on the network I can find all hebi modules with my old laptop but, with the new one, I can only ping them but not visualize them on the Scope.

I also tried the old 1.14 Scope version but I still don't see hebi motors.

Can you please help me with this?

ennerf, 2019-08-09:
Hi,

Scope is not compatible with OpenJDK 11 due to some backwards compatible changes in Java 9, and because it requires JavaFX / OpenJFX.

Please try the standalone version from our [docs page](http://docs.hebi.us/downloads_changelogs.html). It includes an appropriate runtime with all dependencies.

[image]

ennerf, 2019-08-09:
I just saw that the documentation was incorrect.

[image]

I have updated it to the following:

[image]

Giggus, 2019-08-09:
that at least explains why ! Thanks for the reply.. I'll use another laptop for the moment.

Giggus, 2019-08-09:
I actually tried with another Ubuntu 18.04 which has Java 11 and it does work.
I am wondering why on the other platform it doesn't. Are there any other dependencies?

rLinks234, 2019-08-09:
@Giggus , have you done any modifications to your firewall configuration? Even if not, that may still be the issue.

---

Assuming you didn't customize your Ubuntu 18.04 install:

Try disabling your firewall (temporarily) in a console by doing

```sh
sudo ufw disable
```

Make sure to re-enable `ufw` after testing in Scope! (`sudo ufw enable`)

---

If this allows you to see the modules in Scope, then you will need to add a firewall rule to allow traffic from our APIs and Scope:

```sh
sudo ufw allow 16665/udp
```

Giggus, 2019-08-13:
> @Giggus , have you done any modifications to your firewall configuration? Even if not, that may still be the issue.
> 
> Assuming you didn't customize your Ubuntu 18.04 install:
> 
> Try disabling your firewall (temporarily) in a console by doing
> 
> ```shell
> sudo ufw disable
> ```
> 
> Make sure to re-enable `ufw` after testing in Scope! (`sudo ufw enable`)
> 
> If this allows you to see the modules in Scope, then you will need to add a firewall rule to allow traffic from our APIs and Scope:
> 
> ```shell
> sudo ufw allow 16665/udp
> ```

It actually works if I disable the firewall!! But the `sudo ufw allow 16665/udp` doesn't make the job once I re-enable the firewall. 
I'm investigating on it.

Thanks!

rLinks234, 2019-08-13:
I believe that there are additional firewall rules to add to what I mentioned above, @Giggus . It basically comes down to the fact that module discovery requires listening for broadcast packets sent from the modules.

---

To allow broadcast packets in on your network, you will have to add a rule similar to below:

```sh
ufw allow in on <ETH_IFACE> to <ETH_IP_BRDCST> port 16665/udp
```

Above, `ETH_IFACE` is the interface name of your network interface connecting you to the subnet. This is presumably an ethernet - so it may be something like `eno1` or `enp0s31f6`. `ETH_IP_BRDCST` is the **broadcast address of your subnet** - not your network interface's IP address!

---

To determine `ETH_IFACE` from above, you can use the command (in a terminal)

```sh
ip a
```

Using my machine as an example (omitting some data),

```
2: enp0s31f6: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
    inet 10.10.10.xxx/24 brd 10.10.10.255 scope global dynamic noprefixroute enp0s31f6
       valid_lft xxxxxsec preferred_lft xxxxxsec
    inet6 xxxx::xxxx:xxxx:xxx:xxxx/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
```

My broadcast address (`ETH_IP_BRDCST`) is `10.10.10.255` and my `ETH_IFACE` is `enp0s31f6`

---

Note that if you have multiple ethernet interfaces, you will have to make sure that you select the one which connects you to the subnet where your HEBI modules reside.

Please let me know if this works for you.

rLinks234, 2019-08-19:
@Giggus , I apologize for the delay. Try adding these rules to `ufw` and let me know if it resolves your issues:

```sh
sudo ufw allow proto udp from <ETH_SUBNET> to any port 16665
sudo ufw allow proto udp out 16665
```

---

Above, you can determine each `ETH_SUBNET` (you may have more than one if you have multiple ethernet cables/networks on your workstation) by executing `ip a` in a terminal. For example,

```
>>> ip a

... Other interfaces ...

2: enp0s31f6: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
    inet 10.10.10.xxx/24 brd 10.10.10.255 scope global dynamic noprefixroute enp0s31f6
       valid_lft xxxxxsec preferred_lft xxxxxsec
    inet6 xxxx::xxxx:xxxx:xxx:xxxx/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

... Other interfaces ...

```

`ETH_SUBNET` from above would be `10.10.10.0/24`. Most subnets are in the `192.168.x.x` and `10.x.x.x` range, _e.g._:

`10.10.10.0`
`10.10.1.0`
`10.10.2.0`
`192.168.1.0`
`192.168.10.0`

---

Let me know if you have any additional issues.

ennerf, 2019-09-18:
Given that there has been no response I assume that this issue has been resolved.

closed 2019-09-18 by ennerf