HebiRobotics/hebi-scope-feedback #90: Dashed lines are not getting rendered correctly (Scope 1.6)

ennerf, 2019-11-07:
Dashed lines works when points are relatively far away (e.g. fast movements), but they render solid when points are too close to each other.

[image]

RalphSteinhagen, 2019-11-07:
Commit GSI-CS-CO/chart-fx@b45abf7 addresses [Scope #90](https://github.com/HebiRobotics/hebi-scope-feedback/issues/90) and statements like `dataSet1.setStyle("strokeDashPattern= 25, 20, 5, 20;");`being now being correctly passed through to JavaFX's `GraphicsContext` rendering pipe-line.

RalphSteinhagen, 2019-11-07:
A small chart-fx example illustrating a (here deliberately exaggerated) '100-pixel dash, 100-pixel gap' dashed polyline over 1e6 points. The new algorithm draws dashes (when defined) acrossthe individual points and restarts only when NaN values are encountered. 

[image]

N.B. the blue curve has still the error contour surface enabled. Enjoy!

closed 2019-11-11 by ennerf