HebiRobotics/hebi-scope-feedback #98: On plots, provide a buffer to readings that are near min / max

daverolli, 2021-09-17:
Maybe just a 5% - 10% buffer of full scale would work.  That makes it easier to zoom in and see details in context.

ennerf, 2021-09-22:
It should already be doing that. Can you provide more info on where you are encountering problems?

[image]

daverolli, 2021-09-22:
To me, the bottom and top plots above both show the issue. 

 The bottom is readings of the bottom plot are really close to the line.  The top plot gets so close to the top that the hidden ‘zoom’ tools pop down.  I guess what I’m asking for is a bigger buffer?  Or maybe have it based on a percentage of full scale range if it’s not already?

ennerf, 2021-09-22:
I tried to fix this, but ran into a bug https://github.com/GSI-CS-CO/chart-fx/issues/465

ennerf, 2021-09-23:
Found a way to work around it and switched from relative padding to 10% of the full range. How do you like this?

[image]

closed 2021-09-24 by ennerf