HebiRobotics/hebi-scope-feedback #102: Window resize / cropping issue after yellow highlights

daverolli, 2022-05-19:
After getting a box to highlight yellow to indicate a changed value, the window cropping when resizing not longer docks to the top, but becomes centered.  This can hide the top navigation bars on smaller (laptop) screens.

For example, it happens when you change gains for the first time. When the boxes go yellow to highlight changed gains, the windowing changes to being center-crop instead of bottom-crop.  I've seen this both on Windows and macOS.

ennerf, 2022-05-19:
The validator looks for a DecorationPane (StackPane) in the hierarchy and installs one at the root in case it can't find one. Manually adding a DecorationPane beneath the ScrollPane fixes it.

[image]

ennerf, 2022-05-19:
added to 1.7.4

closed 2022-05-19 by ennerf