HebiRobotics/hebi-scope-feedback #104: Scope - Rolling Log

nick-hebi, 2022-08-12:
It would be nice if there was an option for scope to keep a rolling log whenever it is started up. This feature would need to limit the total amount of log files by some predetermined time / file size window. 

This would allow you to always have a log of the last day or so of activity without having to remember to hit the log button every time or to clear old logs manually.

ennerf, 2022-08-12:
For what purpose? Wouldn't this be better done through an API?

It would (1) be really difficult to sift through all that information and (2) would require lots of individual logs because there is no concept of "group".

nick-hebi, 2022-08-12:
The purpose would be mostly to monitor a specific module while doing firmware testing. It would be nice to have logging active all the time in case something unexpected happens. I only really need to monitor one thing at a time - I just want to not have to remember to press the log button every time I open scope.

It doesn't have to be fancy, just an option to always record specific modules. It could just make a new log file every hour and delete anything that is older than 24 hours. I suppose that could be done with API / a script but it seemed like something that might fit in with scope.

ennerf, 2022-08-12:
To me that sounds more like an internal development feature that would be better solved with a commandline tool like `continuous_log  <frequency> <addresses>`.

That way you can put it in auto-start on boot and the frequency won't be impacted by e.g. plotting jitters.

closed 2022-08-19 by nick-hebi