HebiRobotics/hebi-scope-feedback #105: PWM monitoring panel adjustments

iamtesch, 2022-08-19:
From @HEBI-Willig :

It would be cool if an Actuator is in "Direct PWM" that the Scope Monitoring Sliders adjusted accordingly in that Position and Velocity went null or greyed out and the Effort slider actually renamed itself "PWM" and properly plotting command and feedback with the PWM signal to the motor. Just a thought for a nice tool for the future.

nick-hebi, 2022-08-19:
I think there was some minor confusion here on what direct PWM mode does. This mode is open loop so there isn't exactly a feedback value to be plotted. I am not sure what the best way to communicate this through the UI is - either to make the feedback value in the effort plot always match the commanded value or to eliminate the feedback value in the plot entirely.

nick-hebi, 2022-08-19:
I know this is enumerated in the documentation but it might be worth adding some sort of blurb or warning in the UI: 
[image]

iamtesch, 2022-08-19:
My thoughts:

I'm less sure about the command and feedback plots changing here -- the PWM signal to the motor is different than the effort command even in direct PWM, b/c it is going through safety controllers.  Having both is useful

nick-hebi, 2022-08-19:
I think that the current plots are slightly confusing. If you put an actuator in direct PWM mode and command a PWM value it will show effort feedback (in Nm) alongside the PWM value (which is in duty cycle, but labeled as Nm). 

[image]

So in direct PWM mode it might make sense to change the Y axis to duty cycle and plot commanded PWM next to the actual PWM after the safety controllers. Or to disable the effort plot entirely in direct PWM mode and change the PWM plot to include commanded vs actual (safety limited) PWM.

nick-hebi, 2022-08-19:
Also that is an internal actuator with a messed up encoder so the effort readings are wonky - but the UI weirdness should be independent of that

ennerf, 2022-08-19:
Direct PWM already changes the slider to be -1 to +1 for that module. Changing the plot gets really tricky because (among ther reasons) you can have multiple devices with different strategies on a single plot.

HEBI-Willig, 2022-08-19:
Fair points all around. Maybe just the first half at least then for the sliders (position and vel don't do anything anyways). And changing the label on Effort to PWM so it;s not confusing. This is obviously very very low priority as it's mainly an internal control strategy and 99.9% of customers shouldn't be using it.

nick-hebi, 2022-08-19:
> Having two different units makes no sense in any case, so maybe we should just completely disable the effort chart instead?

Yeah I think just disabling the effort chart for modules in direct PWM mode is the best solution. Changing the name of the effort slider / disabling the other sliders would also be a nice to have if it isn't hard to do

daverolli, 2022-12-14:
Thumbs for greying out Position and Velocity sliders when in DIRECT PWM if that's something doable.  It definitely confused a customer recently to the point where they thought what we shipped was not working (and its confused me briefly on a number of occasions).

Leaving the plots as-is for now is ok.  The main thing is to make it clear that the actuator/module is in a mode where it won't respond to anything but commands on the effort channel.

ennerf, 2022-12-14:
[image]

Focusing on a module that is in direct pwm mode now adds a 4th slider for pwm and greys everything else out

ennerf, 2023-02-01:
implemented

closed 2023-02-01 by ennerf