HebiRobotics/hebi-scope-feedback #108: Is there a way to programatically convert .hebilog files to CSV?

mukundbala, 2023-06-06:


ennerf, 2023-06-06:
Which API are you using? In MATLAB there is [HebiUtils::convertGroupLog](https://files.hebi.us/docs/matlab/hebi-matlab-1.8.0/HebiUtils.convertGroupLog.html), but for the others we'd have to add something to the CLI tool.

mukundbala, 2023-06-07:
I am using the python API. My current workflow involves saving .hebilog logs from my robot using the python API. I then convert the .hebilogs manually to CSV using Scope.

A CLI tool would be super helpful!

ennerf, 2023-06-07:
Ok. I just added a conversion tool to the `hebi-config` CLI bundled with [Scope Snapshot](https://hebifiles.z13.web.core.windows.net/download/scope/snapshot/download.html) rev 104 and above:

 ```bash
# convert file as mat / simple to same directory as the original log
hebi-config convertlog -i input.hebilog 

# convert multiple input files to a specified output directory
hebi-config convertlog -i input.hebilog  -i input2.hebilog --output-dir output

# convert to csv / full
hebi-config convertlog -i input.hebilog  --format csv --view full
```

mukundbala, 2023-06-07:
Works perfectly! Thank you @ennerf :)